VOLTTRON/volttron #2818: Volttron Script grab_bacnet_config.py fails with undefined binaryOutput or binaryValue

Bowriverstudio, 2021-10-13:
### Description of Issue
When I run 

`
python3 scripts/bacnet/grab_bacnet_config.py $DEVICE_ID --ini scripts/bacnet/BACpypes.ini --driver-out-file config/mark/devices/$DEVICE_ID --registry-out-file config/mark/registry_configs/$DEVICE_ID.csv --object_device_index=20
`

I expect all points to be found - but it skips over binaryOutput and binaryValue that do not have a description.

### Affected Version

8.1

bonicim, 2021-10-18:
To add on to the root cause of this issue, the script fails to grab any point that does not have a "description" property because the script executes Python2 incompatible code from Bacpypes while using Python3, which is a required by Volttron. Since the script logs the error but does not exit on this failure, the script just ignores the point that it is trying to find and moves on to the next point. 

When @Bowriverstudio ran his script, the logs showed an AttributeError stating the dictionary doesn't have a value called "sort".

```
  File "scripts/bacnet/brs_grab_bacnet_config.py", line 270, in process_object
    Enumerated.keylist(present_value_type(0))
  File "/home/awg_rp/volttron/env/lib/python3.8/site-packages/bacpypes/primitivedata.py", line 1117, in keylist
    items.sort(lambda a, b: self.cmp(a[1], b[1]))
AttributeError: 'dict_items' object has no attribute 'sort'
```

This error occurs because the script  uses an older version of Bacpypes (0.16.7) which can call this line of Python 3 [incompatible code](https://github.com/JoelBender/bacpypes/blob/v0.16.7/py34/bacpypes/primitivedata.py#L1117): 
```python
def keylist(self):
        """Return a list of names in order by value."""
        items = self.enumerations.items()
        items.sort(lambda a, b: self.cmp(a[1], b[1]))
```

The line, ```items.sort```, is no longer valid Python3 code. The script calls that [method in the subroutine 'process_object'](https://github.com/VOLTTRON/volttron/blob/develop/scripts/bacnet/grab_bacnet_config.py#L223). 

```python
# from grab.bacnet_config.py

def process_object(app, address, obj_type, index, max_range_report, config_writer):
        ...
        
       if not object_notes:
            enum_strings = []
            for name in Enumerated.keylist(present_value_type(0)):
```

By calling that method using Python3, the script will trigger an AttributeError because dictionaries no longer have a "sort" method in Python3. [Python3 documentation](https://docs.python.org/3/whatsnew/3.0.html#views-and-iterators-instead-of-lists) states that:

```
dict methods dict.keys(), dict.items() and dict.values() return “views” instead of lists. For example, this no longer works: k = d.keys(); k.sort(). Use k = sorted(d) instead (this works in Python 2.5 too and is just as efficient).
```

In short, until Volttron is updated to use the latest Bacpypes version that is exclusively Python 3 compatible , this script needs to code around this Python 2 backwards incompatibility error. #2820 has been created to update Volttron's version of Bacpypes.

bonicim, 2022-03-16:
Closing since #2819 addressed this issue.

closed 2022-03-16 by bonicim