VOLTTRON/volttron #2862: document rest api group

craig8, 2022-01-29:
https://volttron.readthedocs.io/en/develop/platform-features/web-api/authentication-endpoints.html documents the authentication endpoint.

However how to add a group to a user is undocumented.  Also, vui group is hidden in the bowels of the code. 

I am wondering if we should have an ACL editor or something with volttron.  Thoughts @jhaack, @shwethanidd @schandrika ???

@davidraker Can you take care of this documentation?

davidraker, 2022-01-29:
I can document it.

I've been thinking about this for a while, though, and I really think this needs some type of system to be implemented. Really, there ought to be more fine-grained permissions than just having full access to the API. For instance, we might want a user to be able to use endpoints for devices or history but not to make configuration changes. I haven't implemented anything like that in the API because, while it would be trivial to check more specific claims, it would get cumbersome really quickly if you need to manually add all the groups to the web-users.json file for each user. There should definitely be a config system for adding users and permissions for them. It needn't be specific to the API, either. It might usefully apply to OS users as well.

craig8, 2022-01-29:
I totally agree that there is that need.  And it is general enough of a use case that we can probably justify it with @jhaack.  

Lets create an issue for this as something other than this task and we will discuss it as a team on Monday.