VOLTTRON/volttron #2863: valueerror if Authorization header formatted incorrectly

craig8, 2022-01-29:
A valueerror is thrown on the server if invalid bearer authorization passed (for example no space).  Here is the rfc: https://www.rfc-editor.org/rfc/rfc6750

Anyhow a 500 error is on the client where it should just return unauthorized.

On the server there should be a log about what the value error was however.

This relates to #2860 

```
pattern: ^/vui/platforms/?$, path_info: /vui/platforms
 v: <bound method VUIEndpoints.handle_platforms of <volttron.platform.web.vui_endpoints.VUIEndpoints object at 0x7fb41043f160>>
2022-01-28 19:52:25,512 () volttron.platform.web.platform_web_service DEBUG: registered route t is: callable
Traceback (most recent call last):
  File "/repos/volttron/volttron/platform/web/platform_web_service.py", line 492, in app_routing
    retvalue = v(env, start_response, data)
TypeError: verify_and_dispatch() takes 3 positional arguments but 4 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/repos/volttron/env/lib/python3.8/site-packages/gevent/pywsgi.py", line 970, in handle_one_response
    self.run_application()
  File "/repos/volttron/env/lib/python3.8/site-packages/ws4py/server/geventserver.py", line 63, in run_application
    gevent.pywsgi.WSGIHandler.run_application(self)
  File "/repos/volttron/env/lib/python3.8/site-packages/gevent/pywsgi.py", line 917, in run_application
    self.result = self.application(self.environ, self.start_response)
  File "/repos/volttron/volttron/platform/web/webapp.py", line 38, in __call__
    return self.platformweb.app_routing(environ, start_response)
  File "/repos/volttron/volttron/platform/web/platform_web_service.py", line 494, in app_routing
    response = v(env, data)
  File "/repos/volttron/volttron/platform/web/vui_endpoints.py", line 40, in verify_and_dispatch
    claims = self._agent.get_user_claims(get_bearer(env))
  File "/repos/volttron/volttron/platform/web/__init__.py", line 67, in get_bearer
    auth_type, bearer = http_auth.split(' ')
ValueError: not enough values to unpack (expected 2, got 1)
```

davidraker, 2022-01-30:
Response was changed to 401 Not Authorized for any failed authentication, including invalid tokens. A pull request has been made.

davidraker, 2022-01-30:
Looks like it creates a merge conflict due to other specific exceptions having been added to the same line of code.  My reading of this seems to indicate it should always return 401 if anything goes wrong during authorization, but if you don't agree we can simply add ValueError instead to the list of exceptions.

davidraker, 2022-01-30:
Actually, looks like insufficent claims should result in 403 Forbidden, so I fixed that too, though it is a different line of code just after the except block for everything else.

craig8, 2022-01-31:
I am ok with 403 or 401, but there needs to be a record of what the actual error was within the logs.

davidraker, 2022-01-31:
It is. All failures to authenticate are already logged including the exception message that was raised.The bad token wasn't getting logged only because the ValueError wasn't being caught before.

craig8, 2022-02-01:
yeah but it threw a 500 response error on the server not forbidden.

davidraker, 2022-02-01:
Right, it should now give a 401 or 403 now, as appropriate,  with the pull request, and all occurances are logged as warnings. That last part we could change to info, but it seems like something you might want to track, even with conservative log settings.

closed 2022-02-01 by craig8