VOLTTRON/volttron #2877: Allow volttron to be operated by multiple users in a group

craig8, 2022-02-08:


davidraker, 2022-02-08:
The University of Toledo is having a problem with this on a new system which our IT department manages.  Permitted users on the university domain can each have an account on the computer with a username matching their Active Directory username (set up by Kerberos on first login). VOLTTRON is installed as a system service and runs as its own OS user. Other OS users who should be able to work with it are added to the volttron OS group. A user in the volttron group can set VOLTTRON_HOME=/home/volttron/.volttron and activate the environment in /home/volttron/env, but when they try to run vctl status, they get a permissions error for .volttron/keystores/control.connection/keystore.json because that file was created with permissions 600. Manually setting the permissions for that file to 660 allow vctl status to execute normally. This is going to be an issue, however, for any file in .volttron where the permissions are different for the volttron OS user and the volttron OS group. Ideally, for this use case, if group access is enabled, the user and group permissions should match for any files which volttron creates.

craig8, 2022-02-09:
@schandrika to look at setting up a mask at the operating system level in order to help this.

craig8, 2022-02-09:
Changing group for files, is that a responsibility for volttron?