VOLTTRON/volttron #2903: Add Security Vulnerabilities scan as part of CI pipeline

bonicim, 2022-03-02:
### Description of Issue
As part of the recommendations on security from the SSC team, add Checkmarx action to current CI pipeline so that we can scan for software vulnerabilities in the Volttron source code. Note that I am waiting on the SSC team to give me a Checkmarx account. 

https://github.com/marketplace/actions/checkmarx-cxflow-action
https://stackoverflow.com/questions/42424678/github-integration-with-checkmarx

### Affected Version

8.2

craig8, 2022-03-02:
@bonicim can we actually do this from outside the lab?

bonicim, 2022-03-02:
> @bonicim can we actually do this from outside the lab?

I don't know. The impression that I got from SSC team was that once they get me a Checkmarx account, I should be able to hit the endpoint from Github.