VOLTTRON/volttron #2945: basehistorian's configure() set max_time_publishing and history_limit_days to timedelta instead of float

schandrika, 2022-03-29:
basehistorian's configure() set max_time_publishing and history_limit_days to timedelta instead of float.
init and methods that use these parameters expect it to float and not timedelta

craig8, 2022-03-29:
Can you provide your thoughts on what action you would like to take?  Either float or timedelta, but consistent. Or a this should do this rather than one it does?

@schandrika

schandrika, 2022-03-29:
I think it should be float. That's what is in the documentation and that is how init() sets it. Config store, and hence configure() method was added later. So to keep it consistent with older version and docs it should be float

craig8, 2022-03-29:
I agree make it a float so we should be just casting the timedelta.