VOLTTRON/volttron #2947: Volttron cannot start on Raspbian 10

bonicim, 2022-04-01:
### Description of Issue

Volttron fails to start on Raspbian 10 because [libffi7](https://sourceware.org/libffi/) is not installed on Raspbian 10. Volttron requires cryptography, which requires ssl; and ssl requires libffi7. A similar issue was noted in another repo; see https://github.com/pyca/bcrypt/issues/274

### Affected Version

main

### Steps to Reproduce

Environment:
OS: Raspbian 10 (aka [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/))
Hardware: Raspberry Pi 4

1. Install Volttron (run bootstrap.py)
2. Start Volttron (use ./start-volttron)

Expected output

```
File "/home/developer/volttron-actual/env/lib/python3.9/site-packages/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
from cryptography.hazmat.bindings._constant_time import lib
ImportError: libffi.so.7: cannot open shared object file: No such file or directory
```

### Additional Details

I was able to start Volttron on Raspbian 10 by installing the required dependencies. See scripts below:

```
wget http://ftp.us.debian.org/debian/pool/main/g/gcc-12/gcc-12-base_12-20220319-1_armhf.deb
sudo dpkg -i gcc-12-base_12-20220319-1_armhf.deb
wget http://ftp.us.debian.org/debian/pool/main/g/gcc-12/libgcc-s1_12-20220319-1_armhf.deb
sudo dpkg -i libgcc-s1_12-20220319-1_armhf.deb
wget http://ftp.us.debian.org/debian/pool/main/libf/libffi/libffi7_3.3-6_armhf.deb
sudo dpkg -i libffi7_3.3-6_armhf.deb
```

bonicim, 2022-04-04:
We should update our documentation to explicitly state which OS's we support. A quick search of 'Raspberry' on our readthedocs implies that we do support Raspbian: https://volttron.readthedocs.io/en/develop/introduction/platform-install.html?highlight=raspberry#on-debian-based-systems-and-centos-6-7

awoz, 2022-04-04:
Consider adding "debian bullseye (for Ubuntu 20.04), as demonstrated by VOLTTRON Quick Start.