VOLTTRON/volttron #2976: grab_bacnet_config.py

bbartling, 2022-06-18:
[This commit in develop branch](https://github.com/VOLTTRON/volttron/blob/develop/scripts/bacnet/grab_bacnet_config.py) from Maurice on the VOLTTRON office hours needs to be looked at closely. I think the changes Maurice did so VOLTTRON can pick up BACnet device binaryOutput works like the output from `grab_bacnet_config.py` looks legit, but something is messing up the platform.driver when I use this config file.

[I originally posted this on stackoverflow 2021](https://stackoverflow.com/questions/68776817/volttron-bacnet-discovery-not-all-points-coming-through) that I noticed that the rooftop unit cooling compressor points were never pulled in when running `grab_bacnet_config.py`. Someone on stackoverflow just suggested that I add in these points manually to the registry config file which I do with a: `vctl config edit platform.driver registry_configs/1100.csv`

Paste in these points:

```
Reference Point Name,Volttron Point Name,Units,Unit Details,BACnet Object Type,Property,Writable,Index,Write Priority,Notes
Compressor 1 Command,Compressor 1 Command,Enum,0-1 (default 0),binaryOutput,presentValue,TRUE,3,7,
Compressor 2 Command,Compressor 2 Command,Enum,0-1 (default 0),binaryOutput,presentValue,TRUE,4,7,
Compressor 3 Command,Compressor 3 Command,Enum,0-1 (default 0),binaryOutput,presentValue,TRUE,5,7,
Compressor 4 Command,Compressor 4 Command,Enum,0-1 (default 0),binaryOutput,presentValue,TRUE,6,7,

```

But something is messed up. I've tried restarting the platform.driver and BACnet proxy and I always see this issue:
```
2022-06-18 14:48:13,725 (platform_driveragent-4.0 2999014) __main__ INFO: Starting driver: slipstream_internal/slipstream_hq/1100
2022-06-18 14:48:13,753 (platform_driveragent-4.0 2999014) <stderr> ERROR: Traceback (most recent call last):
2022-06-18 14:48:13,753 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "src/gevent/greenlet.py", line 854, in gevent._gevent_cgreenlet.Greenlet.run
2022-06-18 14:48:13,753 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "/home/geb/.volttron/agents/db5dc5bf-e75f-4ef3-a672-235f23c3f124/platform_driveragent-4.0/platform_driver/driver.py", line 163, in starting
2022-06-18 14:48:13,753 (platform_driveragent-4.0 2999014) <stderr> ERROR:     self.setup_device()
2022-06-18 14:48:13,753 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "/home/geb/.volttron/agents/db5dc5bf-e75f-4ef3-a672-235f23c3f124/platform_driveragent-4.0/platform_driver/driver.py", line 186, in setup_device
2022-06-18 14:48:13,753 (platform_driveragent-4.0 2999014) <stderr> ERROR:     self.interface = self.get_interface(driver_type, driver_config, registry_config)
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "/home/geb/.volttron/agents/db5dc5bf-e75f-4ef3-a672-235f23c3f124/platform_driveragent-4.0/platform_driver/driver.py", line 158, in get_interface
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:     interface.configure(config_dict, config_string)
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "/home/geb/.volttron/agents/db5dc5bf-e75f-4ef3-a672-235f23c3f124/platform_driveragent-4.0/platform_driver/interfaces/bacnet.py", line 80, in configure
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:     self.parse_config(registry_config_str)
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "/home/geb/.volttron/agents/db5dc5bf-e75f-4ef3-a672-235f23c3f124/platform_driveragent-4.0/platform_driver/interfaces/bacnet.py", line 255, in parse_config
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:     register = Register(index,
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:   File "/home/geb/.volttron/agents/db5dc5bf-e75f-4ef3-a672-235f23c3f124/platform_driveragent-4.0/platform_driver/interfaces/bacnet.py", line 68, in __init__
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:     self.python_type = BACNET_TYPE_MAPPING[object_type]
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR: KeyError: 'accumulator'
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR: 2022-06-18T14:48:13Z <Greenlet at 0x7f023393be10: <bound method DriverAgent.starting of <platform_driver.driver.DriverAgent object at 0x7f0234e77850>>(<volttron.platform.vip.agent.core.BasicCore object)> failed with KeyError
2022-06-18 14:48:13,754 (platform_driveragent-4.0 2999014) <stderr> ERROR:
```

When I use Maurices version of `grab_bacnet_config.py` there ALOT more points that come through where it does seem like all of the points, but its almost like something is messed up in the BACnet proxy.

bbartling, 2022-07-01:
This is my registry config for the RTU with the 8.1.3 version of `grab_bacnet_config.py` that isnt picking up Binary Outputs:

```
Reference Point Name,Volttron Point Name,Units,Unit Details,BACnet Object Type,Property,Writable,Index,Write Priority,Notes
Supply Air Temperature Local,Supply Air Temperature Local,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,1,,
Outdoor Air Temperature Local,Outdoor Air Temperature Local,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,2,,
Return Air Temperature,Return Air Temperature,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,3,,
Mixed Air Temperature Local,Mixed Air Temperature Local,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,4,,
Duct Static Pressure Local,Duct Static Pressure Local,inchesOfWater,-5.02 to 5.02,analogInput,presentValue,FALSE,5,,
Control Cabinet Space Temperature,Control Cabinet Space Temperature,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,6,,
IT Closet Space Temperature,IT Closet Space Temperature,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,7,,
Return Air Humidity Local,Return Air Humidity Local,percent,-150.00 to 150.00,analogInput,presentValue,FALSE,8,,
Outdoor Air Relative Humidity Local,Outdoor Air Relative Humidity Local,percent,-150.00 to 150.00,analogInput,presentValue,FALSE,9,,
Supply Fan Speed Command,Supply Fan Speed Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,1,,
Return Fan Speed Output Command,Return Fan Speed Output Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,2,,
Outdoor Air Damper Command,Outdoor Air Damper Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,3,,
ERU Supply Damper Command,ERU Supply Damper Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,4,,
Return Fan offset,Return Fan offset,percent,-150.00 to 150.00 (default 15.0),analogValue,presentValue,FALSE,1,,
Supply Fan P-Gain,Supply Fan P-Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,2,,
Duct Static Pressure Setpoint BAS,Duct Static Pressure Setpoint BAS,inchesOfWater,-5.02 to 5.02 (default 1.0),analogValue,presentValue,FALSE,3,,
Discharge Air Cooling Setpoint BAS,Discharge Air Cooling Setpoint BAS,degreesFahrenheit,-58.00 to 482.00 (default 60.0),analogValue,presentValue,FALSE,4,,
Cooling Capacity Status,Cooling Capacity Status,percent,-150.00 to 150.00 (default 0.0),analogValue,presentValue,FALSE,5,,
Economizer Minimum Position Setpoint BAS,Economizer Minimum Position Setpoint BAS,percent,-150.00 to 150.00 (default 10.0),analogValue,presentValue,FALSE,6,,
Economizer Outdoor Air Enable Setpoint BAS,Economizer Outdoor Air Enable Setpoint BAS,degreesFahrenheit,-58.00 to 482.00 (default 55.0),analogValue,presentValue,FALSE,7,,
Outdoor Air Temperature BAS,Outdoor Air Temperature BAS,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,8,,
Outdoor Air Temperature Active,Outdoor Air Temperature Active,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,9,,
Space Temperature BAS,Space Temperature BAS,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,10,,
Space Temperature Active,Space Temperature Active,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,11,,
Discharge Air Temperature Setpoint Active,Discharge Air Temperature Setpoint Active,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,12,,
Discharge Air Temperature,Discharge Air Temperature,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,13,,
Cooling P Gain,Cooling P Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,14,,
Duct Static Pressure Setpoint Active,Duct Static Pressure Setpoint Active,inchesOfWater,-5.02 to 5.02 (default 0.0),analogValue,presentValue,FALSE,15,,
Econoomizer P Gain,Econoomizer P Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,16,,
Warmup Terminate Setpoint,Warmup Terminate Setpoint,degreesFahrenheit,-58.00 to 482.00 (default 70.0),analogValue,presentValue,FALSE,17,,Warmup will Initiate three degrees below this setpoint
Heating Capacity Primary Status,Heating Capacity Primary Status,percent,-150.00 to 150.00 (default 0.0),analogValue,presentValue,FALSE,18,,
Heating P-Gain,Heating P-Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,19,,
Discharge Air Heating Setpoint BAS,Discharge Air Heating Setpoint BAS,degreesFahrenheit,-58.00 to 482.00 (default 100.0),analogValue,presentValue,FALSE,20,,
Outdoor Air Enthalpy Active,Outdoor Air Enthalpy Active,btusPerPoundDryAir,-98.90 to 98.90 (default 0.0),analogValue,presentValue,FALSE,21,,
Return Air Enthalpy Active,Return Air Enthalpy Active,btusPerPoundDryAir,-98.90 to 98.90 (default 0.0),analogValue,presentValue,FALSE,22,,
Occupancy Request,Occupancy Request,State,State count: 5 (default 1),multiStateValue,presentValue,FALSE,1,,"1=Unoccupied, 2=Occupied Bypass, 3=Occupied Standby, 4=Auto"
Occupancy Status,Occupancy Status,State,State count: 5 (default 1),multiStateValue,presentValue,FALSE,2,,"1=Unoccupied, 2=Occupied Bypass, 3=Occupied Standby, 4=Auto"
Economizer Airside Enable BAS,Economizer Airside Enable BAS,State,State count: 2 (default 1),multiStateValue,presentValue,FALSE,3,,1=Enabled
```

bbartling, 2022-07-01:
This is my registry config for the RTU with the `develop` version of grab_bacnet_config.py that does pick up the Binary Outputs and **_### a lot more points_** come through but errors out on the accumulator issue. Looking at this registry config file I don't see anything about an accumilator associated with Binary Outputs?

[image]

```
Reference Point Name,Volttron Point Name,Units,Unit Details,BACnet Object Type,Property,Writable,Index,Write Priority,Notes
Supply Air Temperature Local,Supply Air Temperature Local,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,1,,
Outdoor Air Temperature Local,Outdoor Air Temperature Local,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,2,,
Return Air Temperature,Return Air Temperature,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,3,,
Mixed Air Temperature Local,Mixed Air Temperature Local,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,4,,
Duct Static Pressure Local,Duct Static Pressure Local,inchesOfWater,-5.02 to 5.02,analogInput,presentValue,FALSE,5,,
Control Cabinet Space Temperature,Control Cabinet Space Temperature,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,6,,
IT Closet Space Temperature,IT Closet Space Temperature,degreesFahrenheit,-58.00 to 482.00,analogInput,presentValue,FALSE,7,,
Return Air Humidity Local,Return Air Humidity Local,percent,-150.00 to 150.00,analogInput,presentValue,FALSE,8,,
Outdoor Air Relative Humidity Local,Outdoor Air Relative Humidity Local,percent,-150.00 to 150.00,analogInput,presentValue,FALSE,9,,
Space Cooling Deviation,Space Cooling Deviation,noUnits,-1000.00 to 1000.00,analogInput,presentValue,FALSE,10,,
Supply Fan Speed Command,Supply Fan Speed Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,1,,
Return Fan Speed Output Command,Return Fan Speed Output Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,2,,
Outdoor Air Damper Command,Outdoor Air Damper Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,3,,
ERU Supply Damper Command,ERU Supply Damper Command,percent,-150.00 to 150.00 (default 0.0),analogOutput,presentValue,FALSE,4,,
Return Fan offset,Return Fan offset,percent,-150.00 to 150.00 (default 15.0),analogValue,presentValue,FALSE,1,,
Supply Fan P-Gain,Supply Fan P-Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,2,,
Duct Static Pressure Setpoint BAS,Duct Static Pressure Setpoint BAS,inchesOfWater,-5.02 to 5.02 (default 1.0),analogValue,presentValue,FALSE,3,,
Discharge Air Cooling Setpoint BAS,Discharge Air Cooling Setpoint BAS,degreesFahrenheit,-58.00 to 482.00 (default 60.0),analogValue,presentValue,FALSE,4,,
Cooling Capacity Status,Cooling Capacity Status,percent,-150.00 to 150.00 (default 0.0),analogValue,presentValue,FALSE,5,,
Economizer Minimum Position Setpoint BAS,Economizer Minimum Position Setpoint BAS,percent,-150.00 to 150.00 (default 10.0),analogValue,presentValue,FALSE,6,,
Economizer Outdoor Air Enable Setpoint BAS,Economizer Outdoor Air Enable Setpoint BAS,degreesFahrenheit,-58.00 to 482.00 (default 55.0),analogValue,presentValue,FALSE,7,,
Outdoor Air Temperature BAS,Outdoor Air Temperature BAS,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,8,,
Outdoor Air Temperature Active,Outdoor Air Temperature Active,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,9,,
Space Temperature BAS,Space Temperature BAS,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,10,,
Space Temperature Active,Space Temperature Active,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,11,,
Discharge Air Temperature Setpoint Active,Discharge Air Temperature Setpoint Active,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,12,,
Discharge Air Temperature,Discharge Air Temperature,degreesFahrenheit,-58.00 to 482.00 (default 0.0),analogValue,presentValue,FALSE,13,,
Cooling P Gain,Cooling P Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,14,,
Duct Static Pressure Setpoint Active,Duct Static Pressure Setpoint Active,inchesOfWater,-5.02 to 5.02 (default 0.0),analogValue,presentValue,FALSE,15,,
Economizer P Gain,Economizer P Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,16,,
Warmup Terminate Setpoint,Warmup Terminate Setpoint,degreesFahrenheit,-58.00 to 482.00 (default 70.0),analogValue,presentValue,FALSE,17,,Warmup will Initiate three degrees below this setpoint
Heating Capacity Primary Status,Heating Capacity Primary Status,percent,-150.00 to 150.00 (default 0.0),analogValue,presentValue,FALSE,18,,
Heating P-Gain,Heating P-Gain,noUnits,-1000.00 to 1000.00 (default 2.0),analogValue,presentValue,FALSE,19,,
Discharge Air Heating Setpoint BAS,Discharge Air Heating Setpoint BAS,degreesFahrenheit,-58.00 to 482.00 (default 100.0),analogValue,presentValue,FALSE,20,,
Outdoor Air Enthalpy Active,Outdoor Air Enthalpy Active,btusPerPoundDryAir,-98.90 to 98.90 (default 0.0),analogValue,presentValue,FALSE,21,,
Return Air Enthalpy Active,Return Air Enthalpy Active,btusPerPoundDryAir,-98.90 to 98.90 (default 0.0),analogValue,presentValue,FALSE,22,,
Economizer Integral Gain,Economizer Integral Gain,noUnits,-1000.00 to 1000.00 (default 0.5),analogValue,presentValue,FALSE,23,,
[... cut ...]
Economizer/DX Low Ambient Lockout,Economizer/DX Low Ambient Lockout,Enum,0-1 (default 0),binaryValue,presentValue,FALSE,6,,
Occupancy Request,Occupancy Request,State,State count: 5 (default 1),multiStateValue,presentValue,FALSE,1,,"1=Unoccupied, 2=Occupied Bypass, 3=Occupied Standby, 4=Auto"
Occupancy Status,Occupancy Status,State,State count: 5 (default 1),multiStateValue,presentValue,FALSE,2,,"1=Unoccupied, 2=Occupied Bypass, 3=Occupied Standby, 4=Auto"
Economizer Airside Enable BAS,Economizer Airside Enable BAS,State,State count: 2 (default 1),multiStateValue,presentValue,FALSE,3,,1=Enabled
```

rlutes, 2022-07-01:
In the bacnet driver interface the appropriate BACnet to python typing translation is defined.  I have not thoroughly reviewed but it might be as simple as adding the following to the dictionary:

"accumulator": int

<code>
BACNET_TYPE_MAPPING = {"multiStateValue": int, "multiStateInput": int, "multiStateOutput": int,
                       "analogValue": float, "analogInput": float, "analogOutput": float,
                       "binaryValue": bool, "binaryInput": bool, "binaryOutput": bool
                      }
<code>

bbartling, 2022-07-01:
Cool. I would be more than happy to test it out if the script can be modified? 

I'm learning a ton about BACnet, I had no idea that there are so many different objects types at least according the bacpypes objects.py

rlutes, 2022-07-01:
Try adding:

"accumulator": int

to the dictionary starting on line 53:

https://github.com/VOLTTRON/volttron/blob/releases/8.1.3/services/core/PlatformDriverAgent/platform_driver/interfaces/bacnet.py

Then you will have to rebuild the platform driver.

bbartling, 2022-07-01:
Got it thanks Robert will test it out soon.

bbartling, 2022-07-02:
So adding this into the PlatformDriverAgent bacnet interface line 53 as you mentioned to do. And using the `develop` version of `grab_bacnet_config.py` it appears to work at no accumulator errors:

[image]

Am hoping to test out writing to these cooling compressor points as well with the actuator agent.

bbartling, 2022-07-06:
@rlutes @jhaack this works. What Robert said to implement on line 53 of the `bacnet.py` of the platform.driver.interfaces I see no errors on my end reading and writing to the Boolean Output cooling compressors on our HVAC system via the actuator agent rpc requests.

From my testing the `develop` branch version of the `grab_bacnet_config.py` works good that Bow Rivers Studio (Maurice from the office hours) merged last fall in conjunction with the line 53 addition for the `bacnet.py` of the platform.driver.interfaces

Thanks for all the help, ill just use these modifications until it can get merged into a non develop git version.