VOLTTRON/volttron #2979: platform driver agent restful interface

bbartling, 2022-07-22:
Update driver to use `grequests` instead of `requests` library.

Currently the driver seems to only support a rest api that has a baseline URL and concatenating the "point" into the url path with no parsing of the payload.

This feature is nice but what if the API doesnt support this and data needs to be parsed? Can an update include optional parsing a payload??

craig8, 2022-07-22:
I assume you are talking about https://github.com/VOLTTRON/volttron/blob/main/services/core/PlatformDriverAgent/platform_driver/interfaces/restful.py.

How would you have the agent parse the payload?  Do you have something in mind?  Will the response be in xml, json, protocol bufs?

I think it might be easier to customize this per the endpoint one is trying to parse.  Then you can supply the logic to parse the data and use the platform driver to format and publish

bbartling, 2022-07-24:
Hi Craig, the only I know how to do is Jerry Rig the `scrape all` to parse everything you need... Like this:

https://gist.github.com/bbartling/9cff12ac5f9f41d0f358b2fec97dcc7c#:~:text=def%20_scrape_all(self)%3A

craig8, 2022-07-25:
That's what you would generally need to do for each of the potential payloads.  I wouldn't call that Jerry Rig though I would call that development using open source technology.  For what you show in your gist is customized to work with what you need it to.

bbartling, 2022-07-25:
Should we be using requests or grequests? Or does it matter?

craig8, 2022-07-28:
If you are going to use it in a greenlet requests is fine, if you are going to use it outside of a greenlet then use grequests.  The difference being that if you are in a greenlet context then you aren't in the main thread of the agent.