VOLTTRON/volttron #2992: weather.gov hourly forecast does not work for lat and lon inputs

schandrika, 2022-09-01:
There is a change in how hourly forecast need to be retrieved from weather.gov api for hourly forecast. latitude and longitude cannot be directly used to retrieve hourly forecast. 
Solution:
When latitude and longitude are given as input, query the api to get the corresponding weather station details and then get hourly forecast using that data

rlutes, 2022-09-01:
call to first:

https://api.weather.gov/points/46.3451,-119.2792

This will then give you:   "forecastHourly": "https://api.weather.gov/gridpoints/PDT/120,133/forecast/hourly",

This is the wfo, x, y needed for the forecast.  It would be great if we could auto-generate the needed wfo, x, and y for both lattitude/long. and weather station ID.

schandrika, 2022-09-01:
@rlutes I fixed this for lat/lon. currently you can send lat/lon or  gridId, gridX, and gridY. Is there a different api for station id?

rlutes, 2022-09-01:
@schandrika if you do a call to stations/{station_id} it returns the lat,long in the geometry dictionary.  Previously, you could do the forecast for lat/long or station ID.

Currently the readme for the agent references the weather station.  This will work for getting the latest observation or reading but not to do the forecast.  Since the main use of the weather agent it to get forecast weather data I think that it would probably be good to update this documentation and include documentation on RPC call to get the hourly forecast.