VOLTTRON/volttron #3011: Test that uses volttron instance fixture cannot connect to config store even though volttron instance is running

bonicim, 2022-09-14:
### Description of Issue

When I try to run an integration test that requires a config store fixture, the test cannot create the fixture because it can't connect to the config store even though the PlatformWrapper (i.e. a test fixture that represents a true, working Volttron platform). 

My fixture errors out on the last line in the code snippet below. I have verified that the platform is indeed running by including an if/else check as seen below. 

```
@pytest.fixture(scope="module")
def config_store_connection(volttron_instance):
    from volttrontesting.utils.platformwrapper import start_wrapper_platform 
    
    if not volttron_instance.is_running():
        start_wrapper_platform(volttron_instance)
    else:
        print("PLATFORM IS ALREADY RUNNING")
    
    capabilities = [{'edit_config_store': {'identity': PLATFORM_DRIVER}}]
    connection = volttron_instance.build_connection(peer=CONFIGURATION_STORE, capabilities=capabilities)

.....
```

Since this test runs two tests on two different volttron instances, I noticed that the errors were different in both of them:

On the first test, the platform appears to be running but the 'serverkey' wasn't set thus not allowing the fixture to establish a connection to the platform:

```
____________________________________________________________________________ ERROR at setup of test_no_groups[volttron_instance0] _____________________________________________________________________________

volttron_instance = volttron_home: /tmp/tmpybl5_5l1/volttron_home

    @pytest.fixture(scope="module")
    def config_store_connection(volttron_instance):
        from volttrontesting.utils.platformwrapper import start_wrapper_platform
    
        if not volttron_instance.is_running():
            start_wrapper_platform(volttron_instance)
        else:
            print("PLATFORM IS ALREADY RUNNING")
    
        capabilities = [{'edit_config_store': {'identity': PLATFORM_DRIVER}}]
>       connection = volttron_instance.build_connection(peer=CONFIGURATION_STORE, capabilities=capabilities)

services/core/PlatformDriverAgent/tests/test_device_groups.py:130: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = volttron_home: /tmp/tmpybl5_5l1/volttron_home, peer = 'config.store', address = 'tcp://127.0.0.102:5841', identity = '15dcb94e-345a-43f5-8f0c-636aaa7ad184', publickey = None, secretkey = None
serverkey = None, capabilities = [{'edit_config_store': {'identity': 'platform.driver'}}], kwargs = {}

    def build_connection(self, peer=None, address=None, identity=None,
                         publickey=None, secretkey=None, serverkey=None,
                         capabilities: Optional[dict] = None, **kwargs):
        self.logit('Building connection to {}'.format(peer))
        os.environ.update(self.env)
        self.allow_all_connections()
    
        if identity is None:
            # Set identity here instead of AuthEntry creating one and use that identity to create Connection class.
            # This is to ensure that RMQ test cases get the correct current user that matches the auth entry made
            identity = str(uuid.uuid4())
        if address is None:
            self.logit(
                'Default address was None so setting to current instances')
            address = self.vip_address
        if self.auth_enabled:
            if address is None:
                serverkey = self.serverkey
            if serverkey is None:
                self.logit("serverkey wasn't set but the address was.")
>               raise Exception("Invalid state.")
E               Exception: Invalid state.

volttrontesting/utils/platformwrapper.py:416: Exception

```

However, on the second test, I see that although the platform is running, the logs show that the connection couldn't be established because the secretkey, publickey, and serverkey were not defined:

```

____________________________________________________________________________ ERROR at setup of test_no_groups[volttron_instance2] _____________________________________________________________________________

volttron_instance = volttron_home: /tmp/tmppnomzv6r/volttron_home

    @pytest.fixture(scope="module")
    def config_store_connection(volttron_instance):
        from volttrontesting.utils.platformwrapper import start_wrapper_platform
    
        if not volttron_instance.is_running():
            start_wrapper_platform(volttron_instance)
        else:
            print("PLATFORM IS ALREADY RUNNING")
    
        capabilities = [{'edit_config_store': {'identity': PLATFORM_DRIVER}}]
>       connection = volttron_instance.build_connection(peer=CONFIGURATION_STORE, capabilities=capabilities)

services/core/PlatformDriverAgent/tests/test_device_groups.py:130: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
volttrontesting/utils/platformwrapper.py:433: in build_connection
    conn = Connection(address=address, peer=peer, publickey=publickey,
volttron/platform/vip/agent/connection.py:97: in __init__
    full_address = build_vip_address_string(
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

vip_root = 'tcp://127.0.0.197:5591', serverkey = None, publickey = None, secretkey = None

    def build_vip_address_string(vip_root, serverkey, publickey, secretkey):
        """ Build a full vip address string based upon the passed arguments
    
        All arguments are required to be non-None in order for the string to be
        created successfully.
    
        :raises ValueError if one of the parameters is None.
        """
        _log.debug("root: {}, serverkey: {}, publickey: {}, secretkey: {}".format(
            vip_root, serverkey, publickey, secretkey))
        parsed = urlparse(vip_root)
        if parsed.scheme == 'tcp':
            if not (serverkey and publickey and secretkey and vip_root):
>               raise ValueError("All parameters must be entered.")
E               ValueError: All parameters must be entered.
```

### Affected Version

releases/8.2

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce
Run pytest below:

` pytest -s services/core/PlatformDriverAgent/tests/test_device_groups.py::test_no_groups`

### Additional Details

craig8, 2022-09-29:
@bonicim  please reference the pr that this was fixed in and then close this issue.

bonicim, 2022-09-30:
The tests that were affected by this issue have been addressed in #3013, #3014, #3015. Instead of using the config_store_connection, the tests use an agent making RPC calls to the Config Store. I suspect that the tests did not create the config_store_connection fixture correctly.

closed 2022-09-30 by bonicim