VOLTTRON/volttron #3066: market isn't working correctly

AntonLED, 2023-01-20:
I'm trying to setup market in my VOLTTRON system. 

For this I use **market_serviceagent-1.0** in _/volttron/services/contrib/MarketServiceAgent_ 
and some example market-agents:
- **meteragent-0.1**
- **electric_buyeragent-0.1**
- **electric_meteragent-0.1**
from _/volttron/examples/MarketAgents._

I don't see any response. 
The main reason, I think, is that the market-service doesn't call the callback functions from market participants code (for example, like on a screenshot)
[image]

Operating System: 
 - ubuntu 22.04 LTS
 - Volttron Version 8.2

**_P.S._**
in the future, I want to set up a market for single board computers (arm architecture), for example Raspberry Pi

AntonLED, 2023-01-20:
I solved this problem by changing market service-agent type from event_driven to periodic:
[image]

closed 2023-01-20 by AntonLED

awoz, 2023-01-20:
Is there a pull request for this fix? Otherwise consider re-opening this issue so that dev team can submit your corrections and test for potential regressions.

craig8, 2023-01-20:
@rlutes have we done anything with the event_driven in a while?  Might be worth looking at.