VOLTTRON/volttron #3077: Infinite log printing at startup

drq, 2023-03-06:
**Describe the bug**
We have installed releases/8.2 with rabbitmq
We are using Python 3.7.1 since we found out Python 3.6.8 doesn't work with the missing dataclasses module.
The OS is Red Hat Enterprise Linux 8.2 (Ootpa)

The installation finished without any error. But when we run the ./start-volttron.

1) -vv is not supported error.

2) if we change -vv to -v or using the rotating log option we see the volttron.log showing millions of lines such as

```
2023-03-06 18:23:44,334 () volttron.platform.auth.auth INFO: auth file /opt/poc/.volttron/auth.json loaded
2023-03-06 18:23:44,334 () volttron.platform.auth.auth INFO: loading auth file /opt/poc/.volttron/auth.json

```
And we have to kill the process since the log will keep growing.

The first a few lines of the log looks like

```
2023-03-06 18:23:05,472 () rmq_setup.py INFO: Rmq server at /opt/poc/rabbitmq_server/rabbitmq_se
rver-3.9.7 is running at
2023-03-06 18:23:05,597 () volttron.platform.store INFO: Initializing configuration store servic
e.
2023-03-06 18:23:05,661 () volttron.platform.auth.auth INFO: loading auth file /opt/poc/.volttro
n/auth.json
2023-03-06 18:23:05,662 () volttron.platform.auth.auth INFO: auth file /opt/poc/.volttron/auth.j
son loaded
2023-03-06 18:23:05,664 () volttron.platform.agent.utils INFO: Adding file watch for /opt/poc/.v
olttron/auth.json dirname=/opt/poc/.volttron, filename=auth.json
2023-03-06 18:23:05,667 () volttron.platform.agent.utils INFO: Added file watch for /opt/poc/.vo
lttron/auth.json
2023-03-06 18:23:05,667 () volttron.platform.agent.utils INFO: Adding file watch for /opt/poc/.v
olttron/protected_topics.json dirname=/opt/poc/.volttron, filename=protected_topics.json
2023-03-06 18:23:05,668 () volttron.platform.agent.utils INFO: Added file watch for /opt/poc/.vo
lttron/protected_topics.json

```

So somewhere in the code, it gets into infinite loop.

Could you help us find out what went wrong?

craig8, 2023-06-01:
This is the same issue that you mention.  I will take a look and see if I can fix it soon.

[image]

craig8, 2023-06-06:
@drq Please use the develop branch for testing if this is fixed for you.

drq, 2023-06-12:
@craig8 Yes. That fixed it.

closed 2023-06-12 by craig8