VOLTTRON/volttron #3086: Setup Platform Driver Config Store: operation timed out

bbartling, 2023-06-03:
I am trying out modular volttron where this works where the all the directions work fine:
https://pypi.org/project/volttron-lib-bacnet-driver/

if I do a vclt status: 
```
$ vctl status
UUID AGENT                             IDENTITY                     TAG     STATUS          HEALTH
9 volttron-bacnet-proxy-0.2.0rc0    platform.bacnet_proxy                running [44133] GOOD
a volttron-listener-0.2.0rc0        volttron-listener-0.2.0rc0_2         running [43496] GOOD
e volttron-openadr-ven-1.0.2rc1     openadr.ven                  openadr
6 volttron-platform-driver-0.2.0rc1 platform.driver                      running [43531] GOOD
```

But having issues trying to setup the Platform Driver configuration store. If I try:
```
$ vctl config store platform.driver registry_configs/10.csv registry_configs/10.csv --csv
config function add_config_to_store: operation timed out
```
This will error out on:
`config function add_config_to_store: operation timed out`
And there is no errors in the log file either.

bonicim, 2023-06-06:
@bbartling

Can you run `vctl config list platform.driver`  to verify that nothing was indeed added to the config store? 

Also, can you verify that the file location of your csv file is correct? That should be the third argument in your command (ie. registry_configs/10.csv).  

On another note, for issues related to modular volttron, please open an issue in the respective repo. For this one, I would recommend putting it in the platform-driver repo: https://github.com/eclipse-volttron/volttron-platform-driver/issues

bonicim, 2023-06-06:
No, the conventions surrounding registry config files from Volttron 7.0 to 8.2 have not changed. You can use old registry files and be able to store them on the config store. 

Can you try installing a registry and device config using the instructions here: https://github.com/eclipse-volttron/volttron-lib-bacnet-driver/tree/develop ? 

I have tried that example to verify that config store works on modular volttron. If you can make it work on your machine, then you can use that same pattern/instructions to store your own bacnet registry configs.

bbartling, 2023-06-07:
Hi @bonicim that does work using the same files as you stated in the README direction from the link you sent.

`bacnet.config` and a `bacnet.csv`

Where then using:
```
vctl config store platform.driver bacnet.csv bacnet.csv --csv
vctl config store platform.driver devices/bacnet bacnet.config
```

Seems to work just fine. Looks like it takes just fine:
```
$ vctl config list platform.driver
bacnet.csv
devices/bacnet
```

I dont remember in 7.0 version when setting up the config store having to have `.config` file extension could that having anything to do with it? My old files dont have a file extensions. Looks like this for example:

[image]

Should I rename all files in my `devices` directory with a `.config` extension?

closed 2023-06-09 by bbartling