VOLTTRON/volttron #3130: Add base historian error handling when device point data has invalid string content

schandrika, 2023-10-20:
User posted html string in a point to device topic instead of to record topic. This cause base_historian to error. The user was testing this in older volttron(7.0.1?) 
Test and make sure historian handles this correctly in the latest volttron version.

schandrika, 2023-10-20:
What should be expected behavior. Should we skip that single point and still record everything else for that timestep and record error? Should status of historian show bad? @rlutes @shwethanidd @craig8 - thoughts?

schandrika, 2023-10-21:
I am not able to reproduce this error in develop, releases/8.2, releases/8.1.

craig8, 2023-10-23:
I would say that the user should update to 8.2.

schandrika, 2023-10-25:
@rlutes could you comment. Could the users update to 8.2?