VOLTTRON/volttron #3147: forward historian sending device data even when configured to only send record data.

ntenney, 2023-12-07:
**Describe the bug**
We have a forwarder configured to only send record data. That forwarder is sending device data when that is configured to not send.

**To Reproduce**

Use the following configuration (with modifications for the address of the VIP and the server key) on a forwardhistorian. Start the VIP and remote platforms and observed the data that is being sent to the VIP.

> destination-serverkey : &lt;server key&gt;
> capture_analysis_data : false,
> capture_record_data : true,
> capture_device_data : false,
> capture_log_data : false,
> destination-vip : tcp://&lt;VIP address&gt;:&lt;vip port&gt;

**Expected behavior**

It is expected that ONLY record data is being sent to the VIP

**Screenshots**

Not a screenshot, but here is an entry in the log file of the VIP that was receiving from the forwarder with the config previously noted:

> 2023-12-01 00:00:00,051 (listeneragent-3.3 37251) __main__ INFO: Peer: pubsub, Sender: bsf.2222.vc.forwarder.record:, Bus: , Topic: devices/PNNL/BSF_CSF/RTU3/FCU13159/all, Headers: {'Date': '20
> 23-12-01T08:03:26.839507+00:00', 'TimeStamp': '2023-12-01T08:03:26.839507+00:00', 'SynchronizedTimeStamp': '2023-12-01T08:03:00.000000+00:00', 'min_compatible_version': '3.0', 'max_compatible_v
> ersion': '', 'X-Forwarded': True, 'X-Forwarded-From': 'bsf.2222'}, Message:
> [{'OccupancyMode': 0,
>   'TerminalBoxFanStatus': 0,
>   'UnoccupiedCoolingTemperatureSetPoint': 76.0,
>   'UnoccupiedHeatingTemperatureSetPoint': 60.0,
>   'ZoneAirFlow': 103.0,
>   'ZoneAirFlowSetPoint': 0.0,
>   'ZoneChilledWaterValvePosition': 0.0,
>   'ZoneCoolingTemperatureSetPoint': 76.0,
>   'ZoneDischargeAirTemperature': 67.9000015258789,
>   'ZoneHeatingTemperatureSetPoint': 60.0,
>   'ZoneHotWaterValvePosition': 0.0,
>   'ZoneTemperature': 61.10000228881836},
>  {'OccupancyMode': {'type': 'float', 'tz': 'US/Pacific', 'units': 'Enum'},
>   'TerminalBoxFanStatus': {'type': 'float',
>                            'tz': 'US/Pacific',
>                            'units': 'Enum'},
>   'UnoccupiedCoolingTemperatureSetPoint': {'type': 'float',
>                                            'tz': 'US/Pacific',
>                                            'units': 'degreesFahrenheit'},
>   'UnoccupiedHeatingTemperatureSetPoint': {'type': 'float',
>                                            'tz': 'US/Pacific',
>                                            'units': 'degreesFahrenheit'},
>   'ZoneAirFlow': {'type': 'float',
>                   'tz': 'US/Pacific',
>                   'units': 'cubicFeetPerMinute'},
>   'ZoneAirFlowSetPoint': {'type': 'float',
>                           'tz': 'US/Pacific',
>                           'units': 'cubicFeetPerMinute'},
>   'ZoneChilledWaterValvePosition': {'type': 'float',
>                                     'tz': 'US/Pacific',
>                                     'units': 'percent'},
>   'ZoneCoolingTemperatureSetPoint': {'type': 'float',
>                                      'tz': 'US/Pacific',
>                                      'units': 'degreesFahrenheit'},
>   'ZoneDischargeAirTemperature': {'type': 'float',
>                                   'tz': 'US/Pacific',
>                                   'units': 'degreesFahrenheit'},
>   'ZoneHeatingTemperatureSetPoint': {'type': 'float',
>                                      'tz': 'US/Pacific',
>                                      'units': 'degreesFahrenheit'},
>   'ZoneHotWaterValvePosition': {'type': 'float',
>                                 'tz': 'US/Pacific',
>                                 'units': 'percent'},
>   'ZoneTemperature': {'type': 'float',
>                       'tz': 'US/Pacific',
>                       'units': 'degreesFahrenheit'}}]
> 

**Operating System (please complete the following information):**
 - OS: Linux/Ubuntu
 - Volttron Version releases/8.2

**Additional context**
Add any other context about the problem here.

ntenney, 2023-12-07:
We discovered the error. The problem was a configuration issue. The config we were using had commas.

closed 2023-12-07 by ntenney

craig8, 2023-12-08:
Well that's good!