VOLTTRON/volttron #3148: Question about debugging VOLTTRON

kefeimo, 2024-01-06:
Background, in the VOLTTRON™ Office Hours - 4/28/2023 (https://www.youtube.com/watch?v=5caMIDeoieo&t=964s) (around: 15:00-20:00) it talks about debugging volttron using vscode. One scenario is when setting debug point at line 435 volttron/platform/main.py, it didn't go to the brakepoint. And the thread issue was mentioned. Could somebody elaborate on that?

kefeimo, 2024-01-06:
Is the launch.py file for this tutorial available somewhere? Especially what is the setup for listener-agent section (around 24:00)?

craig8, 2024-01-06:
I am going to assume you mean launch.json file as it is what vs code uses for launching code..  Debugging volttron.platform.main would require you to start the server in the debugger. 

The following will let you do the server.
```json
...
          {
            "name": "volttron -vv",
            "type": "python",
            "request": "launch",
            "program": "env/bin/volttron",
            "console": "integratedTerminal",
            "justMyCode": true,
            "args": [
                "-vv"
            ],
            "env": {
                "GEVENT_SUPPORT": "True"
            },
            "cwd": "${workspaceFolder}"
        },
...
```

The thread issue is weird.  Basically it has to do with the GEVENT_SUPPORT=True.   That flag allows greenlets to be debugged and have them hit the break points, however it also has the side effect of "Normal threads" not be able to hit break points.  At least that is the observed behavior.   if you remove the GEVENT_SUPPORT then your normal threads will be able to be hit but your greenlets won't be able to be debugged.  I don't think there is any other in-depth discovery of exactly why or how you could do both.