VOLTTRON/volttron #3150: Unable to complete bootstrap.py

flintsaw, 2024-01-31:
**Describe the bug**
After following the [documentation here](https://volttron.readthedocs.io/en/develop/introduction/platform-install.html), I arrive at the error below. I've been able to work around it somewhat but the system does not seem to be functioning properly. For example I can't get `vcfg` to run at all and can't get `vctl` to start any agents.

**Expected behavior**
bootstrap.py completes with no errors

**Actual behavior**
The script hangs here for a bit
[image]

Then fails like this
[image]

**Operating System (please complete the following information):**
Debian GNU/Linux 12 (bookworm)
Volttron develop

craig8, 2024-01-31:
Hmm,  Generally we see this when build-essential is not installed.  Can you test to see if that is installed by running the following command?

```bash
> apt list build-essential
Listing... Done
build-essential/jammy,now 12.9ubuntu3 amd64 [installed]
```

Note if you do not get a line with [installed] then that's the issue.  Please keep us informed!

flintsaw, 2024-01-31:
Can confirm it's installed
`build-essential/stable,now 12.9 amd64 [installed]`

craig8, 2024-01-31:
I forgot to ask if this is a raspberry pi?  Or what is the specs of the system you are working with?  I have a staff member looking into bookworm right now, since its not one that we test natively on.  Our system is Ubuntu based, however debian should just "work" in my opinion as we aren't doing anything extra ordinary!

flintsaw, 2024-01-31:
It's a VM with 2 CPUs, 8GB ram, and a big hard drive. We wanted to stick to the tested distros but long term support for bookworm made the decision easy.

davidraker, 2024-01-31:
The issue here appears to be an incompatibility between some of the required dependencies and Python 3.11 (the default python on Bookworm). Basically, Python moved some of its c headers, and packages which utilize these needed to be updated to work on 3.11. Of our dependencies, this is known to at least have been an issue for pyzmq and gevent. **At this time, VOLTTRON 9.x only supports running on Python versions up to 3.10. The best path to resolve this is to create a virtual environment with Python 3.10. The bootstrap should not then encounter this error.**  

If you are feeling adventuresome, and do not wish to install another python version, you can try changing the version pins in requirements.py for those two packages.  Please understand that we have **not** tested this for the 9.x release, so we can't promise it will work correctly.  You are welcome to try it, but your mileage may vary. For reference, VOLTTRON version 10 (which is still in development but _does_ support running on Python 3.11) uses these package versions:
pyzmq >= "25.0.2"
gevent >= "22.10.2"

flintsaw, 2024-02-02:
No success after changing the versions in requirements.py.
I also can't find a way to get 3.10-venv into Debian 12 though I could be missing something obvious and absolutely welcome advice.
Version 9 is in pre-release, am I missing V10 somewhere?
I'm happy to test anything you like to help development! Unless you recommend another course of action I'll table Volttron for now.

davidraker, 2024-02-02:
The current develop branch of VOLTTRON is version 9. Version 10 will be the next release (in development), and refactors the code into a more modular form.  The code is located here, but will be installed from pypi:

https://github.com/eclipse-volttron

 ------------------------------

I would follow the procedure here to get Python 3.10 on your Debian system (Bookworm is Debian 12, but the procedure will be the same):

https://www.rosehosting.com/blog/how-to-install-python-3-10-on-debian-11/

You probably don't even need to bother with update-alternatives at the end, since you'll be using it in a virtual environment anyway. Once it is there, you can run the bootstrap using python3.10 instead of the system python:

path/to/python3.10 boostrap.py

If bootstrap complains about there being an existing virtual environment, add --force to the end of the command.

craig8, 2024-02-02:
Volttron 10 is being actively developed https://github.com/eclipse-volttron/ 

To install it you can just use pip install volttron from a virtual environment....

https://eclipse-volttron.readthedocs.io/en/latest/ (this also has updated 9.0 documentation)

Not all agents have been ported.  But if you go to pypi and search volttron you will see a bunch of things show up.

closed 2024-02-02 by flintsaw