VOLTTRON/volttron #3187: VOLTTRON: installation error configparser.NoSectionError: No section: 'volttron'

Matammanjunath, 2024-06-24:
**Describe the bug**
I tried installing VOLTTRON but no success. I had installed and used it for a year and it was working fine. Recently, I deleted the VOLTTRON folder and decided to do the installation afresh again. I am surprised to see no success to this time. Don't know why?

**To Reproduce**
```
Steps to reproduce the behavior:
sudo apt-get update 
sudo apt-get full-upgrade 
sudo apt-get install build-essential python3-dev python3-venv openssl libssl-dev libevent-dev git
sudo apt-get install libffi-dev
git clone https://github.com/VOLTTRON/volttron
cd volttron
python3 bootstrap.py 
source env/bin/activate
./start-volttron
```

**Expected behavior**
A clear and concise description of what you expected to happen.
```
pi@raspberrypi:~ $ cd volttron
pi@raspberrypi:~/volttron $ source env/bin/activate
(volttron) pi@raspberrypi:~/volttron $ ./start-volttron
Starting VOLTTRON verbosely in the background with VOLTTRON_HOME=/home/pi/.volttron
Waiting for VOLTTRON to startup..
VOLTTRON startup failed/timed out. Please check volttron.log for details
(volttron) pi@raspberrypi:~/volttron $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
(volttron) pi@raspberrypi:~/volttron $ tail -f volttron.log
    class Agent:
  File "/home/pi/volttron/volttron/platform/vip/agent/__init__.py", line 76, in Agent
    volttron_central_address=None, volttron_central_instance_name=None, enable_auth=is_auth_enabled()):
  File "/home/pi/volttron/volttron/platform/agent/utils.py", line 234, in is_auth_enabled
    config = load_platform_config()
  File "/home/pi/volttron/volttron/platform/agent/utils.py", line 169, in load_platform_config
    options = parser.options('volttron')
  File "/usr/lib/python3.9/configparser.py", line 675, in options
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'volttron'
S^C
(volttron) pi@raspberrypi:~/volttron $ 
```

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Operating System (please complete the following information):**
 - OS: [e.g. iOS]
 - Volttron Version [develop, releases/8.2, main]

**Additional context**
Add any other context about the problem here.

craig8, 2024-06-24:
Did you remove your ~/.volttron home when upgrading?

Matammanjunath, 2024-06-25:
1. I am sure I did not remove any VOLTTRON_HOME when upgrading. Can you please let me know how to do it. In the process of debugging the error, I manually deleted the /home/pi/volttron folder and performed the VOLTTRON installation as per the list of codes given below. This I have done over a dozen times hoping to fix the error. Looks like my following installation instructions do have incomplete steps. Can you please guide me to a list of codes I can execute to install the VOLTTRON successfully. It is hard to find on the VOLTTRON documentation website. 

2. I checked ~/.volttron/config file. It does have following: 
[image]
```
pi@raspberrypi:~ $ cat ~/.volttron/config
[volttron]
message-bus = zmq
vip-address = tcp://127.0.0.1:22916
instance-name = tcp://127.0.0.1:22916
pi@raspberrypi:~ $ 

```

**COMPLETE VOLTTRON INSTALLATION STEPS:** 
Step1: Install prerequisites. Execute the following one by one.

```
sudo apt-get update 
sudo apt-get full-upgrade 
sudo apt-get install build-essential python3-dev python3-venv openssl libssl-dev libevent-dev git
sudo apt-get install libffi-dev

```
Step2: Clone VOLTTRON code. The voltron will download into the home directory.

`git clone https://github.com/VOLTTRON/volttron --branch main`

Step3: Set up virtual environment. Just go for ZMQ bus installation and this is easier and faster to install. I did not install RabbitMQ.

```
cd volttron
python3 bootstrap.py # Coffee is a good to have now. It takes good time. 
source env/bin/activate # activate the volttron platform 

```
Step4: Test the VOLLTRON.
```

# start the volttron
./start-volttron
# vctl stands for volttron control; install an agent (listens to user and controls something) by giving a tag name
vctl install examples/ListenerAgent --tag listener
# now make the agent start/run
vctl start --tag listener
# check the status of listeners 
vctl status
# check the volttron log files
tail -f volttron.log
# to stop the volttron
./stop-volttron
# to deactivate the volttron platform
deactivate

```

craig8, 2024-06-26:
I would go ahead and remove the ~/.volttron directory from your system and then run ./start-volttron

Matammanjunath, 2024-07-08:
Resolved. Removing `~/.volttron` did the trick. It took me a while to figure `.volttron` is a hidden file (DOT in the beginning of the name). After removing it, the installation was smooth and successful. Thanks a ton.

closed 2024-07-08 by Matammanjunath

craig8, 2024-07-08:
Awesome...yeah hidden files so you don't accidentally delete them!  Wouldn't want you to delete it on accident right!