VOLTTRON/volttron #3200: missing keyword on reloader

porkytin, 2024-11-28:
**Describe the bug**
A clear and concise description of what the bug is.

In volttron/utils/__init__.py line 107

```python
super(VolttronHomeFileReloader, self).__init__([get_home() + '/' + filetowatch])
```
The parent class requires the explicit keyword `patterns=[get_home() + '/' + filetowatch]`
This mess ups the vc startup when using http at least.

**To Reproduce**
Steps to reproduce the behavior:
1. install vc using vcfg
2. Start volttron
3. look for error:
```

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 906, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/user/volttron/volttron/platform/web/platform_web_service.py", line 768, in startupagent
    self._admin_endpoints = AdminEndpoints(rpc_caller=rpc_caller)
  File "/home/user/volttron/volttron/platform/web/admin_endpoints.py", line 94, in __init__
    VolttronHomeFileReloader("web-users.json", self.reload_userdict),
  File "/home/user/volttron/volttron/utils/__init__.py", line 109, in __init__
    super(VolttronHomeFileReloader, self).__init__([get_home() + '/' + filetowatch])
TypeError: PatternMatchingEventHandler.__init__() takes 1 positional argument but 2 were given
```

**Operating System (please complete the following information):**
 - OS:  Ubuntu 22.04 on Rpi4
 - Volttron Version [9.0.1]

martindnv, 2024-11-28:
I am using python 3.10 btw

craig8, 2024-12-05:
@porkytin thank you I will get this fixed up today!

craig8, 2025-01-27:
Because we are using python3.8 + we need to pin to less than 5.0 

https://github.com/VOLTTRON/volttron/tree/releases/9.0.3 has this in it.

PR #3206 (2025-01-25): Pinned watchdog to less than version 5.
# Description

The watchdog package added a requirement for keyword arguments in version 5.0.0. The current version watchdog is 6.0.0, so new environments will fail when the VolttronHomeFileReloader or AbsolutePathFileReloader are used. This affects the web API on startup, since VolttronHomeFileReloaders is used for web-users.json.

Unfortunately, simply updating to use keyword arguments also broke other things when running on python 3.8, as watchdog removed support for python 3.8 in the same version. Instead, pinned watchdog version to less than 5.0.0,  [... cut]